ToniRV/NeRF-SLAM #7: Build problem about gtsam

JIANG-CX, 2022-12-28:
When I run the command "cmake --build build_gtsam --config RelWithDebInfo -j", it gave an error as follows. Could you please help me to fix it? Thanks. 

`Traceback (most recent call last):
  File "/home/jiangcx/Documents/Nerf_related/NeRF-SLAM/thirdparty/gtsam/wrap/scripts/pybind_wrap.py", line 94, in <module>
    main()
  File "/home/jiangcx/Documents/Nerf_related/NeRF-SLAM/thirdparty/gtsam/wrap/scripts/pybind_wrap.py", line 90, in main
    wrapper.wrap(sources, args.out)
  File "/home/jiangcx/Documents/Nerf_related/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/pybind_wrapper.py", line 723, in wrap
    submodules=submodules)
  File "/home/jiangcx/Documents/Nerf_related/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/pybind_wrapper.py", line 624, in wrap_file
    module = parser.Module.parseString(content)
  File "/home/jiangcx/Documents/Nerf_related/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py", line 56, in parseString
    return Module.rule.parseString(s)[0]
  File "/opt/ros/noetic/lib/python3/dist-packages/pyparsing/core.py", line 1141, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected string_end, found 'namespace'  (at char 1249), (line:46, col:1)
make[2]: *** [python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/build.make:76: python/gtsam_unstable.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:32620: python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
`

pgaston, 2022-12-28:
Same issue here.    New Ubuntu 20.04 install.

phamquandung, 2022-12-29:
I also have this problem too.

phamquandung, 2022-12-29:
I found the solution. Please go to the folder thirdparty and delete gtsam then git clone the updated (https://github.com/borglab/gtsam). Then build again.

@ToniRV . I think you should update the gtsam in thirdparty to avoid this issue.

hjxwhy, 2022-12-29:
@phamquandung I have done as you said, but I still have the problem. Which version pyparsing you used?

JIANG-CX, 2022-12-29:
> @JIANG-CX, @pgaston I found the solution. Please go to the folder thirdparty and delete gtsam then git clone the updated (https://github.com/ToniRV/gtsam-1/tree/feature/nerf_slam). Then build again.
> 
> @ToniRV . I think you should update the gtsam in thirdparty to avoid this issue.

After downloading the new gtsam,  I find this error still exists.

phamquandung, 2022-12-29:
@JIANG-CX @pgaston @hjxwhy you can fork https://github.com/borglab/gtsam this one and build python wrapper like this https://github.com/borglab/gtsam/tree/develop/python. I tried, and it works.

pgaston, 2022-12-29:
Okay, still hung up here.     I moved the gtsam you specified above and built it per instructions.   However...   when I run I get:

  File "/home/pg/repos/NeRF-SLAM/./examples/../slam/vio_slam.py", line 65, in initial_state
    naive_pose = gtsam.Pose3.identity()
AttributeError: type object 'gtsam.gtsam.Pose3' has no attribute 'identity'

Thoughts?

richard7714, 2022-12-30:
> 

I think it's a problem of gtsam version. According to https://github.com/borglab/gtsam/issues/1298, identity() was replaced to Identity() (UpperCase)

phamquandung, 2022-12-30:
Then, another error: 

File "/home/user/research/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 48, in gtsam_pose_to_torch
    q = pose.rotation().quaternion()
AttributeError: 'gtsam.gtsam.Rot3' object has no attribute 'quaternion'

richard7714, 2022-12-30:
> Then, another error:
> 
> File "/home/user/research/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 48, in gtsam_pose_to_torch q = pose.rotation().quaternion() AttributeError: 'gtsam.gtsam.Rot3' object has no attribute 'quaternion'

That line and the following return statement should be replaced like this
```
    q = pose.rotation().toQuaternion()
    return torch.tensor([t[0], t[1], t[2], q.x(), q.y(), q.z(), q.w()], device=device, dtype=dtype)
```

JIANG-CX, 2022-12-30:
> @JIANG-CX @pgaston @hjxwhy you can fork https://github.com/borglab/gtsam this one and build python wrapper like this https://github.com/borglab/gtsam/tree/develop/python. I tried, and it works.

Thanks for your kindly help. I can pass the build step now. But when I run the "./examples/slam_demo.py”, I find there is an error about the gtsam. I wonder have you met such error? How did you solve it? Thanks a lot.
`  File "./examples/slam_demo.py", line 200, in <module>
    run(args)
  File "./examples/slam_demo.py", line 143, in run
    slam_module.spin() # visualizer should be the main spin, but pytorch has a memory bug/leak if threaded...
  File "./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "./examples/../slam/slam_module.py", line 18, in initialize_module
    from slam.vio_slam import VioSLAM
  File "./examples/../slam/vio_slam.py", line 9, in <module>
    from factor_graph.factor_graph import TorchFactorGraph
  File "./examples/../factor_graph/factor_graph.py", line 14, in <module>
    from gtsam import NonlinearFactorGraph as FactorGraph
ModuleNotFoundError: No module named 'gtsam'
`

phamquandung, 2022-12-30:
> Thanks for your kindly help. I can pass the build step now. But when I run the "./examples/slam_demo.py”, I find there is an error about the gtsam. I wonder have you met such error? How did you [...]

In the build folder you should run make python-install and make python-test, then it should be OK.

JIANG-CX, 2022-12-30:
> 
Yes, I have run make python-install and make python-test, and got the feedback "[100%] Built target python-test" and "Successfully installed gtsam-4.2a8". But the error still exits.

jrpowers, 2023-01-01:
Maybe you installed gtsam to another python environment?

Also, I made a bunch the changes to adopt to new gtsam in this PR, in case it helps:  https://github.com/ToniRV/NeRF-SLAM/issues/12

sxyu, 2023-01-01:
Using the branch `feature/nerf_slam` https://github.com/ToniRV/gtsam-1/tree/feature/nerf_slam
I found that there is a syntax error in `gtsam_unstable/gtsam_unstable.i`
causing a parsing error:
 `virtual class LinearizedHessianFactor : gtsam::LinearizedGaussianFactor {`
Needs to be closed 2 lines below (line 818) with `};`.

yzslab, 2023-01-02:
> Maybe you installed gtsam to another python environment?
> 
> Also, I made a bunch the changes to adopt to new gtsam in this PR, in case it helps: #12

It works well! Nice job!

pgaston, 2023-01-02:
@jrpowers Using your latest PR I got everything to build - whew.    The cube sample scene runs fine!    **_Unfortunately_**, the first example dies a horrible death...    (perhaps I'll wait for the dust to settle on all these changes...)    **Thanks again to all!**

```
(nerf) pg@pg-MS-7A59:~/repos/NeRF-SLAM$ python ./examples/slam_demo.py --dataset_dir=./datasets/Replica/office0 --dataset_name=nerf --buffer=100 --slam --img_stride=2 --fusion='nerf' --multi_gpu --gui
Running with GPUs: 0
Running pipeline in sequential mode.
Loading 1000 images.
  0%|                                                  | 0/1000 [00:00<?, ?it/s]/home/pg/miniconda3/envs/nerf/lib/python3.9/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 1 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
Segmentation fault (core dumped)
```

The "GPUs: 0" looks fishy.    However, nvidia-smi; nvcc --version; and nvitop -- all seem to be showing the usual results.

sxyu, 2023-01-02:
works fully for me with the fixed GTSAM from the author & (only) python changes in this repo made by jrpowers

pengwangucla, 2023-01-05:
what is you gcc version or other environment setup ? , after switch the branch. met this error 

```
/usr/include/c++/8/bits/alloc_traits.h:392:27: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
       using pointer = _Tp*;
                           ^
/usr/include/c++/8/bits/alloc_traits.h:395:39: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
       using const_pointer = const _Tp*;
```
Looks like version issue from gcc 

> works fully for me with the fixed GTSAM from the author & (only) python changes in this repo made by jrpowers

JIANG-CX, 2023-01-06:
> what is you gcc version or other environment setup ? , after switch the branch. met this error
> 
> ```
> /usr/include/c++/8/bits/alloc_traits.h:392:27: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
>        using pointer = _Tp*;
>                            ^
> /usr/include/c++/8/bits/alloc_traits.h:395:39: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
>        using const_pointer = const _Tp*;
> ```
> 
> Solved by checking: [image]
> 
> > works fully for me with the fixed GTSAM from the author & (only) python changes in this repo made by jrpowers

I do  not understand how to change the code in linear.cpp. Could you tell me the what the inner const X& stands for? Thanks.

OceanYing, 2023-01-08:
> > what is you gcc version or other environment setup ? , after switch the branch. met this error
> > ```
> > /usr/include/c++/8/bits/alloc_traits.h:392:27: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
> >        using pointer = _Tp*;
> >                            ^
> > /usr/include/c++/8/bits/alloc_traits.h:395:39: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
> >        using const_pointer = const _Tp*;
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Solved by checking: [image]
> > > works fully for me with the fixed GTSAM from the author & (only) python changes in this repo made by jrpowers
> 
> I do not understand this comment. Could you tell me how to fix the linear.cpp? What's the "inner const X& ..." stands for?

It seems that the solution comes from [README installation-issues: 3](https://github.com/ToniRV/NeRF-SLAM#installation-issues). But i'm also not clear about the "inner const X& ..." and still stuck with this problem now. If you have figured it out, please let me know. @JIANG-CX @pengwangucla

pengwangucla, 2023-01-08:
> 

You may change `const std::vector<const gtsam::Matrix&>&` to `const std::vector<gtsam::Matrix>&`  use & as pointer is not proper define of `std::vector`

OceanYing, 2023-01-09:
> You may change `const std::vector<const gtsam::Matrix&>&` to `const std::vector<gtsam::Matrix>&` use & as pointer is not proper define of `std::vector`

Thanks a lot! That's the solution and it works now!

yusuf-wadi, 2023-04-05:
> After downloading the new gtsam, I find this error still exists.

this worked for me

KnightOfTheMoonlight, 2023-04-17:
> > @phamquandung
> > Then, another error:
> > File "/home/user/research/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 48, in gtsam_pose_to_torch q = pose.rotation().quaternion() AttributeError: 'gtsam.gtsam.Rot3' object has no attribute 'quaternion'
> 
> That line and the following return statement should be replaced like this
> 
> ```
>     q = pose.rotation().toQuaternion()
>     return torch.tensor([t[0], t[1], t[2], q.x(), q.y(), q.z(), q.w()], device=device, dtype=dtype)
> ```

Thanks for your reply! It saved my day! I have run into an error because of wrong modification of this. 

My wrong version is `pose.rotation().toQuaternion().coeffs(); `
This modification leads  q to all `nan`

Tonny24Wang, 2023-07-10:
> When I run the command "cmake --build build_gtsam --config RelWithDebInfo -j", it gave an error as follows. Could you please help me to fix it? Thanks.
>
> `Traceback (most recent call last): File [...]

thirdparty/gtsam/gtsam_unstable/gtsam_unstable.i
  815,1: virtual class LinearizedHessianFactor : gtsam::LinearizedGaussianFactor {
  这个类结尾 少了两个字符  };  ，加上即可

SimonsRoad, 2023-09-04:
try "git clone https://github.com/ToniRV/gtsam-1"
then cmake it!

lllray, 2023-09-24:
> try "git clone https://github.com/ToniRV/gtsam-1" then cmake it!

this worked for me.

wbjsamuel, 2023-12-17:
> @jrpowers Using your latest PR I got everything to build - whew. The cube sample scene runs fine! **_Unfortunately_**, the first example dies a horrible death... (perhaps I'll wait for the dust to [...]

Hi I also met this `segmentation fault` error. Could you please tell me how did you solve this issue? Thx a lot!

Zhangxr02, 2024-05-31:
> Thanks for your kindly help. I can pass the build step now. But when I run the "./examples/slam_demo.py”, I find there is an error about the gtsam. I wonder have you met such error? How did you [...]

Hello, I have the same problem when I run the demo. Although I have update gtsam and torch118, it didn't work. So how did you solve the problem(No module named 'gtsam')?