ToniRV/NeRF-SLAM #8: Nothing to see in the instant-ngp

richard7714, 2022-12-29:
While following the readme, I got this error in the instant-NeRF gui.
[image]

With this error, i also have this error code.
[image]

I tried with gtsam fork, and also with borglab/gtsam, but couldn't fix this problem.

If anyone knows solution, comment me. Thanks.

closed 2022-12-30 by richard7714

phamquandung, 2022-12-30:
Hi @richard7714, can you solve this problem?

richard7714, 2022-12-30:
@phamquandung yes, it was the problem of gtsam/gtsam/linear/linear.i.

Like this comment https://github.com/ToniRV/NeRF-SLAM/issues/5#issuecomment-1367263684, I added `Vector vector(const gtsam::KeyVector& keys) const;` to line 239 of 
thirdparty/gtsam/gtsam/linear/linear.i and add `gtsam::VectorValues optimizeDensely() const;` to 405 of thirdparty/gtsam/gtsam/linear/linear.i.

On top of it, i also deleted `  Vector vector() const;` of line 238 in linear.i

For the gtsam version, I used author's default fork version

phamquandung, 2023-01-05:
Hi @richard7714, even though I fork the new version of gtsam and ngp. I still have this issue. How about your result? 
[image]

richard7714, 2023-01-05:
@phamquandung, Check your terminal and see if there is any error code. I guess it's the problem of your slam module. If it works properly, It goes like this
[image]

songlin, 2023-01-06:
@phamquandung Hi, Do you solve the problem? I might have the same problem as you.  I keep getting 
```
Fusion packet from SLAM module is None...
```
and nothing shows in the GUI.