ToniRV/NeRF-SLAM #10: can NeRF-SLAM run successfully in server

MoonquakesX, 2022-12-30:
Hi! It's a really good job, based on the project description, I'm guessing you're running this project on a desktop, so I'd like to know if it can run on a server, as there may be no desktop visualization on the server, I don't know if this will be a problem, because I prompted an error about X11 when compiling instant-ngp

ToniRV, 2023-01-04:
Hi @MoonquakesX, I guess you have tried running nerf-slam without the `--gui` flag and still failed.
I haven't tried myself, but technically it should be possible to run without display, try perhaps running sigma-fusion instead of nerf-slam to see if it is my code or instant-ngp? In principle, instant-ngp supports headless rendering (https://github.com/NVlabs/instant-ngp/issues/10).

szgy66, 2023-01-11:
> Hi! It's a really good job, based on the project description, I'm guessing you're running this project on a desktop, so I'd like to know if it can run on a server, as there may be no desktop [...]

Do you sloved your problem? I run the sample on the server which without visualization desktop with command:"python ./examples/slam_demo.py --dataset_dir=./datasets/nerf-cube-diorama-dataset/room --dataset_name=nerf --buffer=100 --img_stride=1 --fusion='nerf'" without --gui flag. But when I run, it show:
[image]
Then it does not work continue. So I want to know can your sample run sucessfully without visualization desktop? Thx.