ToniRV/NeRF-SLAM #9: "pyparsing.exceptions.ParseException: Expected string_end, found 'namespace'  (at char 1249), (line:46, col:1) python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/build.make:82: recipe for target 'python/gtsam_unstable.cpp' failed"

rockywind, 2022-12-30:
Hi, 
I met the error bellow when I run the code(cmake --build build_gtsam --config RelWithDebInfo -j).
[image]
[image]

phamquandung, 2022-12-30:
I also have this problem when using your gtsam-1 @ToniRV

jrpowers, 2023-01-01:
I had issues as well (but can't recall if they were the exact same). I worked around them in this PR, could you guys check if this PR works for you?

https://github.com/ToniRV/NeRF-SLAM/issues/12

mysticalfirellama, 2023-01-02:
I am also experiencing this issue when running the code, also at the `cmake --build build_gtsam --config RelWithDebInfo -j` step.

```
Traceback (most recent call last):
  File "/home/user/directed_study/NeRF-SLAM/thirdparty/gtsam/wrap/scripts/pybind_wrap.py", line 94, in <module>
    main()
  File "/home/user/directed_study/NeRF-SLAM/thirdparty/gtsam/wrap/scripts/pybind_wrap.py", line 90, in main
    wrapper.wrap(sources, args.out)
  File "/home/user/directed_study/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/pybind_wrapper.py", line 723, in wrap
    submodules=submodules)
  File "/home/user/directed_study/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/pybind_wrapper.py", line 624, in wrap_file
    module = parser.Module.parseString(content)
  File "/home/user/directed_study/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py", line 56, in parseString
    return Module.rule.parseString(s)[0]
  File "/usr/local/lib/python3.6/dist-packages/pyparsing.py", line 1955, in parseString
    raise exc
  File "/usr/local/lib/python3.6/dist-packages/pyparsing.py", line 3814, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected stringEnd, found 'n'  (at char 1249), (line:46, col:1)
python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/build.make:75: recipe for target 'python/gtsam_unstable.cpp' failed
make[2]: *** [python/gtsam_unstable.cpp] Error 1
CMakeFiles/Makefile2:32339: recipe for target 'python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/all' failed
make[1]: *** [python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```

I made an attempt to checkout and build #12 , that also seemed to retain this same error at this step.

GCChen97, 2023-01-11:
A hack that is useful for me:
directly remove `+stringEnd` at the 48th line of `/media/gccrcv/Data/Opensources/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py`

ptc-lexvandersluijs, 2023-01-13:
> A hack that is useful for me: directly remove `+stringEnd` at the 48th line of `NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py`

That also solves it for me, while attempting to build on Windows.

tfy14esa, 2023-01-28:
> A hack that is useful for me: directly remove `+stringEnd` at the 48th line of `NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py`

Can report that this also worked for me on a headless linux cluster

z619850002, 2023-02-22:
> A hack that is useful for me: directly remove `+stringEnd` at the 48th line of `NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py`
A new problem is faced with such a modification.
/home/kyrie/Documents/Machine_learning/DeepSLAM/NeRF-SLAM/thirdparty/gtsam/wrap/pybind11/include/pybind11/pybind11.h:1596:9:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const pybind11::detail::initimpl::constructor<Args ...>&, const Extra& ...) [with Args = {const std::vector<long unsigned int, tbb::tbb_allocator<long unsigned int> >&, const std::vector<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&> >&, const std::vector<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&, std::allocator<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&> >&, double}; Extra = {pybind11::arg, pybind11::arg, pybind11::arg, pybind11::arg}; type_ = gtsam::HessianFactor; options = {gtsam::GaussianFactor, boost::shared_ptr<gtsam::HessianFactor>}]’
/home/kyrie/Documents/Machine_learning/DeepSLAM/NeRF-SLAM/build_gtsam/python/linear.cpp:400:200:   required from here
/usr/include/c++/5/bits/alloc_traits.h:450:27: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
       using pointer = _Tp*;
                           ^
/usr/include/c++/5/bits/alloc_traits.h:453:39: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
       using const_pointer = const _Tp*;

z619850002, 2023-02-22:
> A hack that is useful for me: directly remove `+stringEnd` at the 48th line of `NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py`

A new problem is met with such a modification.
> /home/kyrie/Documents/Machine_learning/DeepSLAM/NeRF-SLAM/thirdparty/gtsam/wrap/pybind11/include/pybind11/pybind11.h:1596:9:   required from ‘pybind11::class_<type_, options>& [...]
/home/kyrie/Documents/Machine_learning/DeepSLAM/NeRF-SLAM/build_gtsam/python/linear.cpp:400:200:   required from here
/usr/include/c++/5/bits/alloc_traits.h:450:27: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
       using pointer = _Tp*;
                           ^
/usr/include/c++/5/bits/alloc_traits.h:453:39: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
       using const_pointer = const _Tp*;

Wjt-shift, 2023-05-08:
> 

have you sloved it?

Wjt-shift, 2023-05-08:
> > A hack that is useful for me: directly remove `+stringEnd` at the 48th line of `NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py`
> 
> A new problem is met with such a modification.
> 
> > /home/kyrie/Documents/Machine_learning/DeepSLAM/NeRF-SLAM/thirdparty/gtsam/wrap/pybind11/include/pybind11/pybind11.h:1596:9:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const pybind11::detail::initimpl::constructor<Args ...>&, const Extra& ...) [with Args = {const std::vector<long unsigned int, tbb::tbb_allocator >&, const std::vector<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&> >&, const std::vector<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&, std::allocator<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&> >&, double}; Extra = {pybind11::arg, pybind11::arg, pybind11::arg, pybind11::arg}; type_ = gtsam::HessianFactor; options = {gtsam::GaussianFactor, boost::shared_ptrgtsam::HessianFactor}]’
> > /home/kyrie/Documents/Machine_learning/DeepSLAM/NeRF-SLAM/build_gtsam/python/linear.cpp:400:200:   required from here
> > /usr/include/c++/5/bits/alloc_traits.h:450:27: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
> > using pointer = _Tp*;
> > ^
> > /usr/include/c++/5/bits/alloc_traits.h:453:39: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
> > using const_pointer = const _Tp*;

have you sloved this problem?

nypyp, 2023-06-27:
> A new problem is met with such a modification.

find a way to solve these two problem:
```bash
cd thirdparty/
rm -rf gtsam
git clone https://github.com/ToniRV/gtsam-1
mv gtsam-1/ gtsam
```
and then come back to build gtsam, just find it through the blog:
https://blog.csdn.net/m0_59601332/article/details/130524366?ydreferer=aHR0cHM6Ly93d3cuYmluZy5jb20v

finickyDrone, 2024-02-26:
> > > A hack that is useful for me: directly remove `+stringEnd` at the 48th line of `NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py`
> > 
> > 
> > A new problem is met with such a modification.
> > > /home/kyrie/Documents/Machine_learning/DeepSLAM/NeRF-SLAM/thirdparty/gtsam/wrap/pybind11/include/pybind11/pybind11.h:1596:9:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const pybind11::detail::initimpl::constructor<Args ...>&, const Extra& ...) [with Args = {const std::vector<long unsigned int, tbb::tbb_allocator >&, const std::vector<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&> >&, const std::vector<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&, std::allocator<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&> >&, double}; Extra = {pybind11::arg, pybind11::arg, pybind11::arg, pybind11::arg}; type_ = gtsam::HessianFactor; options = {gtsam::GaussianFactor, boost::shared_ptrgtsam::HessianFactor}]’
> > > /home/kyrie/Documents/Machine_learning/DeepSLAM/NeRF-SLAM/build_gtsam/python/linear.cpp:400:200:   required from here
> > > /usr/include/c++/5/bits/alloc_traits.h:450:27: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
> > > using pointer = _Tp*;
> > > ^
> > > /usr/include/c++/5/bits/alloc_traits.h:453:39: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
> > > using const_pointer = const _Tp*;
> 
> find a way to solve these two problem without change the code:
> 
> ```shell
> cd thirdparty/
> rm -rf gtsam
> git clone https://github.com/ToniRV/gtsam-1
> mv gtsam-1/ gtsam
> ```
> 
> and then come back to build gtsam, and done without erro, find the way through the blog: https://blog.csdn.net/m0_59601332/article/details/130524366?ydreferer=aHR0cHM6Ly93d3cuYmluZy5jb20v

This fixed the issue me. Thanks!