ToniRV/NeRF-SLAM #11: the ngp error in running

chengYi-xun, 2022-12-31:
thank for your work! i have build the project successfully . but when i run the slam_demo with Replica data ,i get some problem.

`Process Process-2:
Traceback (most recent call last):
  File "/home/alis/.conda/envs/nerf-slam/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/alis/.conda/envs/nerf-slam/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/media/alis/EXTERNAL_USB/kitti/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/media/alis/EXTERNAL_USB/kitti/NeRF-SLAM/./examples/../fusion/fusion_module.py", line 25, in initialize_module
    self.fusion = NerfFusion(self.name, self.args, self.device)
  File "/media/alis/EXTERNAL_USB/kitti/NeRF-SLAM/./examples/../fusion/nerf_fusion.py", line 65, in __init__
    self.ngp = ngp.Testbed(mode,0) # NGP can only use device = 0
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. pyngp.Testbed(arg0: pyngp.TestbedMode)
    2. pyngp.Testbed(arg0: pyngp.TestbedMode, arg1: str, arg2: str)
    3. pyngp.Testbed(arg0: pyngp.TestbedMode, arg1: str, arg2: json)
`
when i change the code to self.ngp = ngp.Testbed(mode),this error disappear,but i get new problem 
`Process Process-2:
Traceback (most recent call last):
  File "/home/alis/.conda/envs/nerf-slam/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/alis/.conda/envs/nerf-slam/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/media/alis/EXTERNAL_USB/kitti/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/media/alis/EXTERNAL_USB/kitti/NeRF-SLAM/./examples/../fusion/fusion_module.py", line 25, in initialize_module
    self.fusion = NerfFusion(self.name, self.args, self.device)
  File "/media/alis/EXTERNAL_USB/kitti/NeRF-SLAM/./examples/../fusion/nerf_fusion.py", line 72, in __init__
    self.ngp.create_empty_nerf_dataset(n_images, nerf_scale, offset, aabb_scale, render_aabb)
TypeError: create_empty_nerf_dataset(): incompatible function arguments. The following argument types are supported:
    1. (self: pyngp.Testbed, n_images: int, aabb_scale: int = 1, is_hdr: bool = False) -> None
`
how can i solve this problem ?  i hold you can give me some advice , i couldn't find the reason for the error, and I searched for it for a long time

chengYi-xun, 2022-12-31:
it seem some error in the pyngp ,but i build successfully in the build_ngp and get 100% in the end . It is worth noting that when I am in the terminal `(nerf-slam) alis@alis-System-Product-Name:/media/alis/EXTERNAL_USB/kitti/NeRF-SLAM$ python
>>> import pyngp
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pyngp'
`
can you help me solve this problem ,i can`t run the code successfully

chengYi-xun, 2023-01-01:
i have solve this problem by use feature/nerf_slam branch[https://github.com/ToniRV/NeRF-SLAM/issues/3](url)

closed 2023-01-01 by chengYi-xun

MoonquakesX, 2023-01-01:
@chengYi-xun  Can I ask if you compiled successfully on a server or desktop? Thank you！

xiehousen, 2023-01-04:
> i have solve this problem by use feature/nerf_slam branch #3

It does not works for me. Do you have any idea about this problem?