ToniRV/NeRF-SLAM #14: vio-slam

liangyongshi, 2023-01-03:
I0103 15:19:20.003431 146533 pipeline_module.py:85] Module: VioSLAM - Spinning. Segmentation fault (core dumped)

ToniRV, 2023-01-04:
Ah! Can you provide a stack trace? perhaps with gdb?

liangyongshi, 2023-01-04:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，已经收到您的邮件，无法亲自回复，还请见谅。

liangyongshi, 2023-01-05:
ubuntu 20.04, torch 12.1,
CUDA 11.6,  1 RTX3090,
I have only one GPU, so modify the code as the picture
[image]
run the code
python ./examples/slam_demo.py --dataset_dir=./Datasets/Replica/office0 --dataset_name=nerf --buffer=100  --weights=./droid.pth --img_stride=2 --fusion='nerf'  --slam --gui --parallel_run

results
[image]

liangyongshi, 2023-01-05:
when I close the parallel_run, the result is
[image]

Looking forward to your reply

ZhenhuiL1n, 2023-01-08:
I am also facing this problem, did anyone solve it yet?

ZhenhuiL1n, 2023-01-08:
[image]

ZhenhuiL1n, 2023-01-08:
[image]

liangyongshi, 2023-01-08:
> I am also facing this problem, did anyone solve it yet?

something wrong with torch_scatter and droid_backends,  adopt conda to insatll torch_scatter, and I don't know how to solve the other problem of droid_backends

ZhenhuiL1n, 2023-01-08:
> > I am also facing this problem, did anyone solve it yet?
> 
> something wrong with torch_scatter and droid_backends, adopt conda to insatll torch_scatter, and I don't know how to solve the other problem of droid_backends

I think it should be a specific version, cause I used the conda to install troch_scatter and other bug pops up

phamquandung, 2023-01-09:
For me, I use two GPU and the problem is solved

closed 2023-01-09 by liangyongshi

parastoobakhtiarii, 2025-01-14:
> > I am also facing this problem, did anyone solve it yet?
> 
> something wrong with torch_scatter and droid_backends, adopt conda to insatll torch_scatter, and I don't know how to solve the other problem of droid_backends

Hi, I am trying to run demo, but it seems like the droid-backends is a module specified for this nerf-slam which is some how missed and the setup.py can't find it. Do you know how I can handle this problem? 
Thank you.