ToniRV/NeRF-SLAM #15: question about run demo

qingyuemeng, 2023-01-03:
Hi, thanks for your amazing work,

when I ran the demo use the command
```
python ./examples/slam_demo.py --dataset_dir=./datasets/Replica/office0 --dataset_name=nerf --buffer=100 --slam --parallel_run --img_stride=2 --fusion='nerf' --multi_gpu --gui
```
I got the a lot log like this:
```
Fusion packet from SLAM module is None...
```
and then it miss in fusion loop, is that right?

xiehousen, 2023-01-05:
The same question. Have you solved this problem?

qingyuemeng, 2023-01-05:
> The same question. Have you solved this problem?

@xiehousen not yet😭

Wjt-shift, 2023-01-07:
Have you solved this problem?

szgy66, 2023-02-03:
Have you solved this problem? I meet the same problem, I guess it's because the number of GPU is less than 2. But I can't verify that. Because my lab server does not have a visual desktop, the laptop cannot be installed with a GPU。

rwn17, 2023-04-14:
Hi all, this is just a bug for logging. In my case, the system still works even with this warning.