ToniRV/NeRF-SLAM #17: GUI result does not show anythin

phamquandung, 2023-01-04:
Hi contributors @ToniRV, @jrpowers,

I build the code successfully. However, when I try to run the code as python ./examples/slam_demo.py --dataset_dir=./datasets/Replica/office0 --dataset_name=nerf --buffer=100 --slam --parallel_run --img_stride=2 --fusion='nerf' --gui. I could not see any gui result. I wonder how to solve this problem. Before that, I also saw an error. Does it matter? 
[image]
[image]

bkhanal-11, 2023-01-04:
Ther is a new PR #12 with this isse and others, I was also facing the same issue but using this [fork](https://github.com/jrpowers/NeRF-SLAM) worked for me.

ToniRV, 2023-01-04:
Yes, that seems like the instant-ngp is the one from nvidia instead of my fork + branch `feature/nerf-slam'

songlin, 2023-01-06:
@bkhanal-11 Hi, can you tell me which and how many GPUs you used to run this project?

bkhanal-11, 2023-01-06:
@songlin I am using 2x NVIDIA RTX 3090.

songlin, 2023-01-09:
@bkhanal-11 Thanks for your information. I managed to run it on 1 NVIDIA RTX 3090.