ToniRV/NeRF-SLAM #18: NANs in L_inv!!

Wjt-shift, 2023-01-04:
When I run the code,I got this error.It seems L matrix can't inverse.How should I solve it?Thanks a lot. @ToniRV @jrpowers 
`NANs in L_inv!!
Traceback (most recent call last):
  File "/home/wjt/data/code/NeRF-SLAM/./examples/slam_demo.py", line 200, in <module>
    run(args)
  File "/home/wjt/data/code/NeRF-SLAM/./examples/slam_demo.py", line 179, in run
    and (not slam or slam_module.spin()) \
  File "/home/wjt/data/code/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 101, in spin
    output = self.spin_once(input);
  File "/home/wjt/data/code/NeRF-SLAM/./examples/../slam/slam_module.py", line 11, in spin_once
    output = self.slam(input)
  File "/home/wjt/data/software/anaconda3/envs/nerfslam/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wjt/data/code/NeRF-SLAM/./examples/../slam/meta_slam.py", line 30, in forward
    output = self._frontend(batch["data"], self.state, self.delta)
  File "/home/wjt/data/code/NeRF-SLAM/./examples/../slam/vio_slam.py", line 114, in _frontend
    x0_visual, visual_factors, viz_out = self.visual_frontend(batch)  # TODO: currently also calls BA, and global BA
  File "/home/wjt/data/software/anaconda3/envs/nerfslam/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/wjt/data/code/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 326, in forward
    self.__initialize()
  File "/home/wjt/data/code/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 650, in __initialize
    x0, rcm_factor = self.update(kf0=None, kf1=None, use_inactive=True)
  File "/home/wjt/data/software/anaconda3/envs/nerfslam/lib/python3.9/site-packages/torch/amp/autocast_mode.py", line 12, in decorate_autocast
    return func(*args, **kwargs)
  File "/home/wjt/data/code/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 439, in update
    x0, rcm_factor = self.ba(gru_estimated_flow, gru_estimated_flow_weight, damping,
  File "/home/wjt/data/code/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 1176, in ba
    raise
RuntimeError: No active exception to reraise
`

ToniRV, 2023-01-04:
Does it always happen?
Is it at the beginning of the run?

ChenHoy, 2023-01-14:
Hi, I am experiencing the same issue. This happens consistently when I run the demo code on the example replica scene. 

Any update on this?

ChenHoy, 2023-02-08:
Hi, did anyone resolve this?

lihaowei1999, 2024-06-11:
Maybe you check if the stereo option in arg is True. This happens in my code when slam is using stereo=True with single view dataset.  Change it to false and everything will be good.