ToniRV/NeRF-SLAM #19: The demo result doesn't look so good as the result in paper?

xiehousen, 2023-01-05:
I run the data set of sample provided by `download_replica_sample.bash`, which is the display effect. From this result, it seems that the effect is not as good as that in the paper. Need to change some parameters?

[image]

xiehousen, 2023-01-05:
**_PS:_** WARNING  Could not initialize Vulkan and NGX. DLSS not supported.

phamquandung, 2023-01-05:
Hi @xiehousen. Could you please show me how to get that result? I even cannot get anything.
[image]
[image]

xiehousen, 2023-01-05:
In fact, I don't know how to compile to display. You may need to replace an instant ngp and gtsam version.

In addition, I will display many flags when running. But I don't know whether it is right to output these things.

[image]

phamquandung, 2023-01-05:
I can't see anything in the result
[image]

xiehousen, 2023-01-05:
The `eval` result (PSNR, L1) is  zero.

phamquandung, 2023-01-05:
> The `eval` result (PSNR, L1) is zero.

Same as my case. That's why the output result of GUI is as my aforementioned figure. I wonder what's wrong with this code? @ToniRV @jrpowers

sjdingcn, 2023-06-16:
I got the same problem before and solved it by installing Vulkan SDK and enabling DLSS.
- Install Vulkan SDK: https://vulkan.lunarg.com/doc/view/latest/linux/getting_started_ubuntu.html
- Enable DLSS https://github.com/NVlabs/instant-ngp/issues/621#issuecomment-1133095007

Augusthyq, 2024-01-07:
> > The `eval` result (PSNR, L1) is zero.
> 
> Same as my case. That's why the output result of GUI is as my aforementioned figure. I wonder what's wrong with this code? @ToniRV @jrpowers

I got the same problem.Do you solve it?