ToniRV/NeRF-SLAM #20: Segmentation fault (core dump)

phamquandung, 2023-01-05:
Could you please help me with this issue? @ToniRV @jrpowers 
When I try to run the example: `python ./examples/slam_demo.py --dataset_dir=./datasets/Replica/office0 --dataset_name=nerf --buffer=100 --slam --parallel_run --img_stride=2 --fusion='nerf' --gui`

It turns out that: "Module: VioSLAM - Spinning. Segmentation fault (core dumped)" as 
below figure
[image]

After that, the code continues running as following figure.
[image]
 Finally, the result shows nothing.
[image]

lvmingzhe, 2023-01-08:
Hi, I have the same problem as you.
I follow the [instruction ](https://github.com/jrpowers/NeRF-SLAM)from @jrpowers and it really helps. In the beginning, everything goes fine.
When it comes to the Run part, things get worse.
It turns out that: "Module: VioSLAM - Spinning. Segmentation fault (core dumped).
Have you solved this problem yet?

phamquandung, 2023-01-09:
> Hi, I have the same problem as you. I follow the [instruction ](https://github.com/jrpowers/NeRF-SLAM)from @jrpowers and it really helps. In the beginning, everything goes fine. When it comes to the [...]

I then use 2 GPUs and the problem is solved.

closed 2023-01-10 by phamquandung

apeach64, 2023-02-23:
> I then use 2 GPUs and the problem is solved.

Hello, I had the same problem(core dumped). too. 
I tried with 2 GPUs, but I faced same error.

How can I fix this error? Is there anyone who solve this problem?!

lvmingzhe, 2023-02-27:
No. My problem still exists, and I have to give it up.

apeach64, 2023-03-02:
I solved the problem with reinstall package named 'torch-scatter' after upgrading pip.  

`pip install --upgrade pip setuptools wheel
pip uninstall torch-scatter
pip install torch-scatter`

I hope this comment helps someone.

EdwardjkFeng, 2023-03-17:
Hello @apeach64, I've try uninstall and reinstall torch-scatter, but the problem remains. Can you please provide which version of `torch-scatter` you use?
Besides, my device has only one GPU. I'm curious if you can still run the demo using one GPU.

apeach64, 2023-03-21:
Hi @EdwardjkFeng, my torch-scatter version is 2.1.0
Also I still using NeRF-SLAM with one GPU.
I'll provide some information about my environment in case it might be helpful.

My environment spec is 

> GeForce RTX 3090
> ubuntu 18.04
> cuda 11.7 + cudnn 8.7.0

I hope you can solve the problem.