ToniRV/NeRF-SLAM #21: type object 'gtsam.gtsam.Pose3' has no attribute 'identity'

szgy66, 2023-01-05:
First of all, thank you for publicizing your project.
When I run the command： python. / examples/slam_demo py -- dataset_dir =. / datasets/up/office0 - dataset_name = nerf - buffer = 100 - slam --parallel_run --img_stride=2 --fusion='nerf' --multi_gpu --gui

The following error will occur:

Running with GPUs: 0,1
Running pipeline in parallel mode.
I0105 19:11:41.273782 58239 pipeline_module.py:85] Module: VioSLAM - Spinning.
Traceback (most recent call last):
  File "./examples/slam_demo.py", line 200, in <module>
    run(args)
  File "./examples/slam_demo.py", line 143, in run
    slam_module.spin() # visualizer should be the main spin, but pytorch has a memory bug/leak if threaded...
  File "/workspace/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/workspace/NeRF-SLAM/./examples/../slam/slam_module.py", line 19, in initialize_module
    self.slam = VioSLAM(self.name, self.args, self.device)
  File "/workspace/NeRF-SLAM/./examples/../slam/vio_slam.py", line 81, in __init__
    world_T_imu_t0, _, _ = initial_state()
  File "/workspace/NeRF-SLAM/./examples/../slam/vio_slam.py", line 65, in initial_state
    naive_pose = gtsam.Pose3.identity()
AttributeError: type object 'gtsam.gtsam.Pose3' has no attribute 'identity'

I'm running your project on a server without a visual interface, so I guess it doesn't matter. Can you help me with this problem? Thanks!

bkhanal-11, 2023-01-05:
You can use this [fork](https://github.com/jrpowers/NeRF-SLAM) to recompile NeRF-SLAM. It fixes issues with new version of GTSAM. I did the same and worked for me.

szgy66, 2023-01-06:
> You can use this [fork](https://github.com/jrpowers/NeRF-SLAM) to recompile NeRF-SLAM. It fixes issues with new version of GTSAM. I did the same and worked for me.

Hello, when I recompile the relevant file with your fork. appear this error:
CMakeFiles/Makefile2:32332: recipe for target 'python/CMakeFiles/gtsam_py.dir/all' failed
make[1]: *** [python/CMakeFiles/gtsam_py.dir/all] Error 2
Makefile:165: recipe for target 'all' failed
make: *** [all] Error 2

The same error occurred when I compiled the previous version. pass order: "ti  go it clone https://github.com/ToniRV/gtsam-1.git" solve it.

Now I find that your fork is an improvement on the original version, mainly cuda version and gtsam.
But there are still problems. How can I solve it? If it is resolved in the same way, I think there will still be problems raised by this issue.

bkhanal-11, 2023-01-06:
I check my version of gtsam and the commit of branch used is f427b0411. So, could you do `git checkout f427b0411` for gtsam? I am also testing the code myself so do not know much. Hope this solves your issue.

szgy66, 2023-01-06:
> f427b0411

> I check my version of gtsam and the commit of branch used is f427b0411. So, could you do `git checkout f427b0411` for gtsam? I am also testing the code myself so do not know much. Hope this solves your issue.

Are you sure your branch number is f427b0411? I get nothing.
[image]

And I compare two branches to see what’s changed, don't find  f427b0411.

543630836, 2023-01-06:
I met the same problem with you. Has this problem been solved? Thx.

543630836, 2023-01-06:
> 

I used your gtsam version and recompiled, but still got the same error.

543630836, 2023-01-06:
> Try [this](https://github.com/jrpowers/gtsam/tree/more_python_exports).

I used your gtsam version and recompiled, but still got the same error.

bkhanal-11, 2023-01-06:
@543630836 If you are getting `type object 'gtsam.gtsam.Pose3' has no attribute 'identity'` error, then the problem is with NeRF-SLAM not GTSAM.  

Try cloning [this](https://github.com/jrpowers/NeRF-SLAM) and follow the install instruction [here](https://github.com/jrpowers/NeRF-SLAM/blob/master/README.md).

543630836, 2023-01-06:
> @543630836 If you are getting `type object 'gtsam.gtsam.Pose3' has no attribute 'identity'` error, then the problem is with NeRF-SLAM not GTSAM.
>
> Try cloning [...]

Thanks for your reply.
However, when I run "cmake --build build_ngp --config RelWithDebInfo -j"
The following error will occur:
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/testbed.dir/build.make:118: testbed] Error 1
make[1]: *** [CMakeFiles/Makefile2:204: CMakeFiles/testbed.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX shared library pyngp.cpython-37m-x86_64-linux-gnu.so
[100%] Built target pyngp
make: *** [Makefile:91: all] Error 2

szgy66, 2023-01-09:
> 

Thank you for your advice, I have compiled successfully, there are no gtsam related errors. But when I run demo's command: "./examples/slam_demo.py --dataset_dir=./Datasets/Replica/office0 --dataset_name=nerf --buffer=100 --slam --parallel_run --img_stride=2 --fusion='nerf' --multi_gpu --gui", It has the following error:

Traceback (most recent call last):
  File "/opt/conda/envs/NeRF_SLAM/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/opt/conda/envs/NeRF_SLAM/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/workspace/NeRF_SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/workspace/NeRF_SLAM/./examples/../datasets/data_module.py", line 37, in initialize_module
    self.dataset = NeRFDataset(self.args, self.device)
  File "/workspace/NeRF_SLAM/./examples/../datasets/nerf_dataset.py", line 18, in __init__
    self.parse_metadata()
  File "/workspace/NeRF_SLAM/./examples/../datasets/nerf_dataset.py", line 39, in parse_metadata
    with open(os.path.join(self.dataset_dir, "transforms.json"), 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: './Datasets/Replica/office0/transforms.json'

I downloaded the related data set from README.md, but I did not find the related transforms.json file. Do you have any suggestions. Thanks a lot.

szgy66, 2023-01-09:
> @543630836 If you are getting `type object 'gtsam.gtsam.Pose3' has no attribute 'identity'` error, then the problem is with NeRF-SLAM not GTSAM.
>
> Try cloning [...]

Thank you for your advice, I have compiled successfully, there are no gtsam related errors. But when I run demo's command: "./examples/slam_demo.py --dataset_dir=./Datasets/Replica/office0 --dataset_name=nerf --buffer=100 --slam --parallel_run --img_stride=2 --fusion='nerf' --multi_gpu --gui", It has the following error:

Traceback (most recent call last):
File "/opt/conda/envs/NeRF_SLAM/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
self.run()
File "/opt/conda/envs/NeRF_SLAM/lib/python3.8/multiprocessing/process.py", line 108, in run
self._target(*self._args, **self._kwargs)
File "/workspace/NeRF_SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
self.initialize_module()
File "/workspace/NeRF_SLAM/./examples/../datasets/data_module.py", line 37, in initialize_module
self.dataset = NeRFDataset(self.args, self.device)
File "/workspace/NeRF_SLAM/./examples/../datasets/nerf_dataset.py", line 18, in init
self.parse_metadata()
File "/workspace/NeRF_SLAM/./examples/../datasets/nerf_dataset.py", line 39, in parse_metadata
with open(os.path.join(self.dataset_dir, "transforms.json"), 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: './Datasets/Replica/office0/transforms.json'

I downloaded the related data set from README.md, but I did not find the related transforms.json file. Do you have any suggestions. Thanks a lot.

liangyongshi, 2023-01-09:
I tried all the methods mentioned above, but the same problem

szgy66, 2023-01-09:
> I tried all the methods mentioned above, but the same problem
you can try this :https://github.com/ToniRV/NeRF-SLAM/issues/21#issuecomment-1373536626. It does work for me.

szgy66, 2023-01-09:
> @543630836 Make sure you have extracted the dataset in correct folder. The dataset might be extracted to `./datasets` rather than `./Datasets`.
> 
> Try:
> 
> ```shell
> ./examples/slam_demo.py --dataset_dir=./datasets/Replica/office0 --dataset_name=nerf --buffer=100 --slam --parallel_run --img_stride=2 --fusion='nerf' --multi_gpu --gui
> ```
I changed the file decompressing path and sent the Replica to datasets, but there was still no transforms.json file.
May I ask if this file is generated externally, not as COLMAP or other
[image]

bkhanal-11, 2023-01-11:
@szgy66 I am using x11 forwarding from server for visualization.

XinyueZ, 2023-01-16:
Any progress on this issue? (dont wanna another fork, sorry guys).  It was really frustrating that after a long time install process and finally nothing worked. :-(

XinyueZ, 2023-01-16:
@ToniRV can you help us?

szgy66, 2023-01-17:
> 

To be clear, following @bkhanal-11 fork works just fine. I guess your git instruction is :"git clone https://github.com/ToniRV/NeRF-SLAM.git --recurse-submodules". Its wrong. You should use:  "git clone https://github.com/jrpowers/NeRF-SLAM.git --recurse-submodules". The rest of the operation can follow [this](https://github.com/jrpowers/NeRF-SLAM).
The git address of the first step must be correct. Hope that can help you.

szgy66, 2023-01-17:
> @ToniRV can you help us?

To be clear, following @bkhanal-11 fork works just fine. I guess your git instruction is :"git clone https://github.com/ToniRV/NeRF-SLAM.git --recurse-submodules". Its wrong. You should use: "git clone https://github.com/jrpowers/NeRF-SLAM.git --recurse-submodules". The rest of the operation can follow [this](https://github.com/jrpowers/NeRF-SLAM).
The git address of the first step must be correct. Hope that can help you.

XinyueZ, 2023-01-17:
> To be clear, following @bkhanal-11 fork works just fine. I guess your git instruction is :"git clone https://github.com/ToniRV/NeRF-SLAM.git --recurse-submodules". Its wrong. You should use: "git [...]

thanks @szgy66 , just for g++, which version should be changed to? Currently, I am working with <= 11, is that sufficient? Because I got an error while building instant-ngp.

szgy66, 2023-01-18:
> thanks @szgy66 , just for g++, which version should be changed to? Currently, I am working with <= 11, is that sufficient? Because I got an error while building instant-ngp.

[image]

stopwithfailure, 2023-02-20:
you can use pip install gtsam==4.2a9 to solve all problems about gtsam
if there is no gtsam==4.2a9 you can change the pip source to   https://pypi.mirrors.ustc.edu.cn/simple/

liangyongshi, 2023-02-20:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，已经收到您的邮件，无法亲自回复，还请见谅。

EOF101110, 2023-12-18:
> [image]

I meet a problem：ImportError: /home/mei/anaconda3/envs/nerf-slam2/lib/python3.10/site-packages/torch/lib/../../../../libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/mei/python/NeRF-SLAM/build_ngp/pyngp.cpython-310-x86_64-linux-gnu.so) how can I solve this?

iDonghq, 2024-02-29:
> @543630836 If you are getting `type object 'gtsam.gtsam.Pose3' has no attribute 'identity'` error, then the problem is with NeRF-SLAM not GTSAM.
>
> Try cloning [...]

Not using NeRF-SLAM, but the error is same

WhoKilledTheNorns, 2024-08-09:
Solved  it.
just modify` slam/visual_frontends/visual_frontend.py ` as below:`# def gtsam_pose_to_torch(pose: gtsam.Pose3, device, dtype):
#     t = pose.translation()
#     # q = pose.rotation().quaternion()
#     q = pose.rotation().matrix()
#     return torch.tensor([t[0], t[1], t[2], q[1], q[2], q[3], q[0]], device=device, dtype=dtype)
def gtsam_pose_to_torch(pose: gtsam.Pose3, device, dtype):
    t = pose.translation()
    R_matrix = pose.rotation().matrix()  # 3x3 rotating matrix
    r = R.from_matrix(R_matrix)
    q = r.as_quat()  # transfer to  [qx, qy, qz, qw]
`
remember to import like this:`from scipy.spatial.transform import Rotation as R`