ToniRV/NeRF-SLAM #22: How to use my own RGB photos?

yzslab, 2023-01-07:
Hi, 
I want to use some other real world scene to test the performance of NeRF-SLAM.
I have some RGB photos taken by mobile phone (no camera poses, no depth map). What should I do so that I can put these photos into NeRF-SLAM to do reconstruction?

Thanks.

szgy66, 2023-01-18:
> Hi, I want to use some other real world scene to test the performance of NeRF-SLAM. I have some RGB photos taken by mobile phone (no camera poses, no depth map). What should I do so that I can put these photos into NeRF-SLAM to do reconstruction?
> 
> Thanks.

Hi，Did you have solved?

qungvinh1280, 2023-02-20:
> > Hi, I want to use some other real world scene to test the performance of NeRF-SLAM. I have some RGB photos taken by mobile phone (no camera poses, no depth map). What should I do so that I can put these photos into NeRF-SLAM to do reconstruction?
> > Thanks.
> 
> Hi，Did you have solved?

Hi, I have the same problem, can you guide me how to solve it?