ToniRV/NeRF-SLAM #23: make error when installing gtsam

OceanYing, 2023-01-08:
Thanks for the inspiring work and release of code. I met a problem when making install gtsam in step:
`cmake --build build_gtsam --config RelWithDebInfo -j`
The gtsam branch I used here is the newest one with the correction proposed by [sxyu](https://github.com/ToniRV/NeRF-SLAM/issues/7#issuecomment-1368541680).

The reported error is:

`SLAM/NeRF-SLAM/build_gtsam/python/linear.cpp:401:200:   required from here
/usr/include/c++/7/bits/allocator.h:113:26: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
       typedef _Tp*       pointer;
                          ^~~~~~~
/usr/include/c++/7/bits/allocator.h:114:26: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
       typedef const _Tp* const_pointer;
                          ^~~~~~~~~~~~~
`

which seems the same as reported by [pengwangucla](https://github.com/ToniRV/NeRF-SLAM/issues/7#issuecomment-1371841065). 

After checking , I think the error occurs in linear.cpp (line 401)
`.def(py::init<const gtsam::KeyVector&, const std::vector<const gtsam::Matrix&>&, const std::vector<const gtsam::Vector&>&, double>(), py::arg("js"), py::arg("Gs"), py::arg("gs"), py::arg("f"))`

which is highly related to [what ToniRV has said in README](https://github.com/ToniRV/NeRF-SLAM#installation-issues):

> 3. Somehow the parser converts [this](https://github.com/borglab/gtsam/compare/develop...ToniRV:gtsam-1:feature/nerf_slam#diff-add3627555fb7411e36ea4d863c15f4187e018b6e00b608ab260e3221aef057aR345) to const std::vector<const gtsam::Matrix&>&, and I need to remove manually in gtsam/build/python/linear.cpp the inner **const X& ...**, and also add <pybind11/stl.h> because:
>   Did you forget to `#include <pybind11/stl.h>`?

But I cannot find out the correct way to fix this error. Maybe anyone could help explain **`const X& ...`**  here.
Hope someone could help me out. Thanks a lot!

OceanYing, 2023-01-09:
[pengwangucla's answer](https://github.com/ToniRV/NeRF-SLAM/issues/7#issuecomment-1374904112) helps me out!

> You may change `const std::vector<const gtsam::Matrix&>&` to `const std::vector<gtsam::Matrix>&` since use & as pointer is not proper define of std::vector

So I execute the following command and get the error metioned above:
`cmake --build build_gtsam --config RelWithDebInfo -j`

And now, line 401 in SLAM/NeRF-SLAM/build_gtsam/python/linear.cpp should be changed from:
`.def(py::init<const gtsam::KeyVector&, const std::vector<const gtsam::Matrix&>&, const std::vector<const gtsam::Vector&>&, double>(), py::arg("js"), py::arg("Gs"), py::arg("gs"), py::arg("f"))`
to
`.def(py::init<const gtsam::KeyVector&, const std::vector<gtsam::Matrix>&, const std::vector<gtsam::Vector>&, double>(), py::arg("js"), py::arg("Gs"), py::arg("gs"), py::arg("f"))`

Then execute again
`cmake --build build_gtsam --config RelWithDebInfo -j`

I found it passed now!

JayChen1996, 2023-02-06:
The file "linear.cpp" is auto-generated, what is the root cause?

ArghyaChatterjee, 2023-03-17:
After changing the line, here is another error that I am encountering. Can anybody help ?
```
Traceback (most recent call last):
  File "/home/arghya/NeRF-SLAM/thirdparty/gtsam/wrap/scripts/pybind_wrap.py", line 94, in <module>
    main()
  File "/home/arghya/NeRF-SLAM/thirdparty/gtsam/wrap/scripts/pybind_wrap.py", line 90, in main
    wrapper.wrap(sources, args.out)
  File "/home/arghya/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/pybind_wrapper.py", line 721, in wrap
    cc_content = self.wrap_file(content,
  File "/home/arghya/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/pybind_wrapper.py", line 624, in wrap_file
    module = parser.Module.parseString(content)
  File "/home/arghya/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py", line 56, in parseString
    return Module.rule.parseString(s)[0]
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 1947, in parseString
    raise exc
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 1937, in parseString
    loc, tokens = self._parse(instring, 0)
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 1838, in _parseCache
    value = self._parseNoCache(instring, loc, doActions, callPreParse)
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 1677, in _parseNoCache
    loc, tokens = self.parseImpl(instring, preloc, doActions)
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 4052, in parseImpl
    loc, exprtokens = e._parse(instring, loc, doActions)
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 1838, in _parseCache
    value = self._parseNoCache(instring, loc, doActions, callPreParse)
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 1681, in _parseNoCache
    loc, tokens = self.parseImpl(instring, preloc, doActions)
  File "/usr/lib/python3/dist-packages/pyparsing.py", line 3798, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected stringEnd, found 'n'  (at char 1249), (line:46, col:1)
make[2]: *** [python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/build.make:76: python/gtsam_unstable.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:32601: python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```

yusuf-wadi, 2023-04-05:
Same issue
```bash
Traceback (most recent call last):
  File "/home/yusuf/nerf/NeRF-SLAM/thirdparty/gtsam/wrap/scripts/pybind_wrap.py", line 94, in <module>
    main()
  File "/home/yusuf/nerf/NeRF-SLAM/thirdparty/gtsam/wrap/scripts/pybind_wrap.py", line 90, in main
    wrapper.wrap(sources, args.out)
  File "/home/yusuf/nerf/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/pybind_wrapper.py", line 721, in wrap
    cc_content = self.wrap_file(content,
  File "/home/yusuf/nerf/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/pybind_wrapper.py", line 624, in wrap_file
    module = parser.Module.parseString(content)
  File "/home/yusuf/nerf/NeRF-SLAM/thirdparty/gtsam/wrap/gtwrap/interface_parser/module.py", line 56, in parseString
    return Module.rule.parseString(s)[0]
  File "/home/yusuf/miniconda3/envs/slam/lib/python3.10/site-packages/pyparsing/core.py", line 1141, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected string_end, found 'namespace'  (at char 1249), (line:46, col:1)
gmake[2]: *** [python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/build.make:76: python/gtsam_unstable.cpp] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:32340: python/CMakeFiles/pybind_wrap_gtsam_unstable.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....

```

yusuf-wadi, 2023-04-05:
[https://github.com/ToniRV/gtsam-1/tree/feature/nerf_slam](this)
^^ worked for me, an updated gtsam. delete the current gtsam dir in thirdparty and clone the new instead.