ToniRV/NeRF-SLAM #30: AttributeError: module 'droid_backends' has no attribute 'reduced_camera_matrix'

ZZy129326999, 2023-01-26:
AttributeError: module 'droid_backends' has no attribute 'reduced_camera_matrix'

ZZy129326999, 2023-01-26:
`            H, v, Q, E, w = droid_backends.reduced_camera_matrix(
                self.cam0_T_world,
                self.world_T_body, # TODO(remove, unnecessary, given the math)
                self.cam0_idepths,
                self.cam0_intrinsics[0],
                self.cam0_T_body,
                self.cam0_idepths_sensed,
                gru_estimated_flow, 
                gru_estimated_flow_weight, # TODO: we should pass as well previous pose covariances!, so to not lose information when optimizing
                damping,
                ii, jj, kf0, kf1)
`

ZZy129326999, 2023-01-26:
`            H, v, Q, E, w = droid_backends.reduced_camera_matrix(
                self.cam0_T_world,
                self.world_T_body, # TODO(remove, unnecessary, given the math)
                self.cam0_idepths,
                self.cam0_intrinsics[0],
                self.cam0_T_body,
                self.cam0_idepths_sensed,
                gru_estimated_flow, 
                gru_estimated_flow_weight, # TODO: we should pass as well previous pose covariances!, so to not lose information when optimizing
                damping,
                ii, jj, kf0, kf1)`

ZZy129326999, 2023-01-26:
```
            H, v, Q, E, w = droid_backends.reduced_camera_matrix(
                self.cam0_T_world,
                self.world_T_body, # TODO(remove, unnecessary, given the math)
                self.cam0_idepths,
                self.cam0_intrinsics[0],
                self.cam0_T_body,
                self.cam0_idepths_sensed,
                gru_estimated_flow, 
                gru_estimated_flow_weight, # TODO: we should pass as well previous pose covariances!, so to not lose information when optimizing
                damping,
                ii, jj, kf0, kf1)
```

yonasteodros, 2023-06-24:
@ZZy129326999 Did you manage to fix this error? I am having the same problem. I checked droid_slam/droid_backend.py and I couldn't find this function.

josyulakrishna, 2024-08-04:
> @ZZy129326999 Did you manage to fix this error? I am having the same problem. I checked droid_slam/droid_backend.py and I couldn't find this function.

Hi, Have you solved this? can you please let me know?