ToniRV/NeRF-SLAM #39: TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:

Yuea922, 2023-02-21:
Process Process-2:
Traceback (most recent call last):
  File "/home/qzy/miniconda3/envs/nerf-slam/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/qzy/miniconda3/envs/nerf-slam/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/qzy/1_learn/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/home/qzy/1_learn/NeRF-SLAM/./examples/../fusion/fusion_module.py", line 25, in initialize_module
    self.fusion = NerfFusion(self.name, self.args, self.device)
  File "/home/qzy/1_learn/NeRF-SLAM/./examples/../fusion/nerf_fusion.py", line 65, in __init__
    self.ngp = ngp.Testbed(mode, 0) # NGP can only use device = 0
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. pyngp.Testbed(arg0: pyngp.TestbedMode)
    2. pyngp.Testbed(arg0: pyngp.TestbedMode, arg1: str, arg2: str)
    3. pyngp.Testbed(arg0: pyngp.TestbedMode, arg1: str, arg2: json)

Invoked with: <TestbedMode.Nerf: 0>, 0
Traceback (most recent call last):
  File "./examples/slam_demo.py", line 200, in <module>
    run(args)
  File "./examples/slam_demo.py", line 143, in run
    slam_module.spin() # visualizer should be the main spin, but pytorch has a memory bug/leak if threaded...
  File "/home/qzy/1_learn/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/home/qzy/1_learn/NeRF-SLAM/./examples/../slam/slam_module.py", line 19, in initialize_module
    self.slam = VioSLAM(self.name, self.args, self.device)
  File "/home/qzy/1_learn/NeRF-SLAM/./examples/../slam/vio_slam.py", line 100, in __init__
    self.backend = iSAM2()
  File "/home/qzy/1_learn/NeRF-SLAM/./examples/../solvers/nonlinear_solver.py", line 51, in __init__
    isam_params.enableRelinearization = True
AttributeError: 'gtsam.gtsam.ISAM2Params' object has no attribute 'enableRelinearization'

ZuoJiaxing, 2023-05-06:
How to fix this? I got the same issue

ERGOWHO, 2023-09-22:
I also have TypeError: __init__(): incompatible constructor arguments.  but after this, it still running to 100% and looks like it is training right now, not sure this incompatible constructor arguments is harmful or not

ERGOWHO, 2023-09-22:
> I also have TypeError: **init**(): incompatible constructor arguments. but after this, it still running to 100% and looks like it is training right now, not sure this incompatible constructor arguments is harmful or not

https://github.com/NVlabs/instant-ngp/issues/1209  this one can fix maybe, this is issue with ngp side

HaFred, 2023-11-12:
Using instant-ngp with neither Feb commit nor right now commit can make it work. Anybody got any thoughts on this?