ToniRV/NeRF-SLAM #41: Not able building gtsam - make: *** [Makefile:166: all] Error 2

0x434D, 2023-02-27:
Hey guys,

I keep trying to build gtsam like described in the repo but can't get it to work.

I'm very new to anything c/c++ and cmake related so any help would be highly appreciated !!

OS: Ubuntu 20.04

command --> cmake ./thirdparty/gtsam -DGTSAM_BUILD_PYTHON=1 -B build_gtsam
```
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM Version: 4.2a8
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen version: 3.3.7
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found Python3: /home/chris/Programs/miniconda3/envs/nerf-slam/bin/python3.8 (found version "3.8.16") found components: Interpreter Development Development.Module Development.Embed 
-- Found TBB: /usr/include (found suitable version "2020.1", minimum required is "4.4") found components: tbb tbbmalloc 
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.2.0
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.2.0
-- Install prefix: /usr/local
-- Found Python: /home/chris/Programs/miniconda3/envs/nerf-slam/bin/python3.8 (found suitable exact version "3.8.16") found components: Interpreter Development Development.Module Development.Embed 
[... 138 lines cut ...]
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/chris/repos/NeRF-SLAM/build_gtsam/CMakeFiles/CMakeScratch/TryCompile-onaQU5

Run Build Command(s):/usr/bin/make -f Makefile cmTC_df07c/fast && /usr/bin/make  -f CMakeFiles/cmTC_df07c.dir/build.make CMakeFiles/cmTC_df07c.dir/build
make[1]: Entering directory '/home/chris/repos/NeRF-SLAM/build_gtsam/CMakeFiles/CMakeScratch/TryCompile-onaQU5'
Building C object CMakeFiles/cmTC_df07c.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create -o CMakeFiles/cmTC_df07c.dir/CheckFunctionExists.c.o -c /home/chris/repos/NeRF-SLAM/build_gtsam/CMakeFiles/CMakeScratch/TryCompile-onaQU5/CheckFunctionExists.c
Linking C executable cmTC_df07c
/opt/cmake-3.25.2-linux-x86_64/bin/cmake -E cmake_link_script CMakeFiles/cmTC_df07c.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create -rdynamic CMakeFiles/cmTC_df07c.dir/CheckFunctionExists.c.o -o cmTC_df07c  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_df07c.dir/build.make:99: cmTC_df07c] Error 1
make[1]: Leaving directory '/home/chris/repos/NeRF-SLAM/build_gtsam/CMakeFiles/CMakeScratch/TryCompile-onaQU5'
make: *** [Makefile:127: cmTC_df07c/fast] Error 2
```

0x434D, 2023-02-27:
Maybe to add to it:

I tried deleting the build_gtsam folder completely and redoing the steps to get it 

This results in cmake failing at random percentages - e.g. sometimes it crashes at 17%, 97% or like the one I posted at 58%

Eunjooo, 2023-03-08:
I'm having the same problem, has this been resolved?

1005452649, 2023-04-21:
I'm having the same problem, has this been resolved?