ToniRV/NeRF-SLAM #42: the gt-pose use in instant ngp

chengYi-xun, 2023-03-02:
thanks for your work , in the euroc dataset ,when i use gt_pose replace the driod-slam pose .the mapping is wrong,how can i fix my wrong ,how the gt pose change can work? when i use the gt_pose to map ,the map is as follow 
[image]

chengYi-xun, 2023-03-02:
i hope some one can help me

lihaowei1999, 2024-06-11:
[image]

Line 562 do a inv to the pose. I think you can try inverse the gt pose in previous.