ToniRV/NeRF-SLAM #44: Can't open file .obj with blender or meshlab

qungvinh1280, 2023-03-17:
I am using the NeRF-SLAM source code from this Github link: https://github.com/jrpowers/NeRF-SLAM, and when training, I am using the GUI of instant-ngp. The issue is that when I export the .obj file, I am unable to open it with Blender or meshlab.

L0510410, 2023-03-30:
Probably it‘s because all vertices are inf in the .obj file. If it's the case, you can try modify nerf_fusion.py line 70 and line71 (offset & render_aabb) to some value instead of inf.

qungvinh1280, 2023-04-08:
> Probably it‘s because all vertices are inf in the .obj file. If it's the case, you can try modify nerf_fusion.py line 70 and line71 (offset & render_aabb) to some value instead of inf.

Thank you so much, that is exactly the issue I've been encountering. I have a question, though: are there any rules for replacing 'inf' values? If possible, could you give me an example?