ToniRV/NeRF-SLAM #46: ImportError: /usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch_backends.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK2at10TensorBase8data_ptrIdEEPT_v

1005452649, 2023-04-25:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../gui/gui_module.py", line 19, in initialize_module
    from gui.open3d_gui import Open3dGui
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../gui/open3d_gui.py", line 4, in <module>
    from fusion.tsdf_fusion import TsdfFusion
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../fusion/tsdf_fusion.py", line 6, in <module>
    from lietorch import SE3
  File "/usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch/__init__.py", line 2, in <module>
    from .groups import LieGroupParameter, SO3, RxSO3, SE3, Sim3, cat, stack
  File "/usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch/groups.py", line 5, in <module>
    from .group_ops import Exp, Log, Inv, Mul, Adj, AdjT, Jinv, Act3, Act4, ToMatrix, ToVec, FromVec
  File "/usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch/group_ops.py", line 1, in <module>
    import lietorch_backends
ImportError: /usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch_backends.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK2at10TensorBase8data_ptrIdEEPT_v
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../fusion/fusion_module.py", line 21, in initialize_module
    from fusion.tsdf_fusion import TsdfFusion
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../fusion/tsdf_fusion.py", line 6, in <module>
    from lietorch import SE3
  File "/usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch/__init__.py", line 2, in <module>
    from .groups import LieGroupParameter, SO3, RxSO3, SE3, Sim3, cat, stack
  File "/usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch/groups.py", line 5, in <module>
    from .group_ops import Exp, Log, Inv, Mul, Adj, AdjT, Jinv, Act3, Act4, ToMatrix, ToVec, FromVec
  File "/usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch/group_ops.py", line 1, in <module>
    import lietorch_backends
ImportError: /usr/local/lib/python3.8/dist-packages/droid_backends-0.0.0-py3.8-linux-x86_64.egg/lietorch_backends.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK2at10TensorBase8data_ptrIdEEPT_v

1005452649, 2023-04-26:
have solved this problem myself

Hezhexi2002, 2023-04-26:
> 

Hi,I meet a similar problem with you:
[image]
could you please tell me how you solved this error finally?

1005452649, 2023-04-26:
Check whether the cuda and pytorch versions match, and then update pytorch to match the cuda version @Hezhexi2002

1005452649, 2023-04-26:
but i have meet  problem now 
AttributeError: 'gtsam.gtsam.GaussianFactorGraph' object has no attribute 'optimizeDensely'

Hezhexi2002, 2023-04-26:
> Check whether the cuda and pytorch versions match, and then update pytorch to match the cuda version @Hezhexi2002

[image]
the verion of my cuda and torch should be matched like the screenshot shown above,but the error still exits

1005452649, 2023-04-26:
your cuda version? @Hezhexi2002

Hezhexi2002, 2023-04-26:
> your cuda version? @Hezhexi2002

[image]
my cuda version is 11 as you can see,so you mean I must reinstall the torch with cu118?

Hezhexi2002, 2023-04-26:
> > your cuda version? @Hezhexi2002
> 
> [image] my cuda version is 11.8 as you can see,so you mean I must reinstall the torch with cu118?

[image]
[image]
as you can see,the error still exits after I install the torch exactly match my cuda version,is it related to my gtsam installation but the compilation is completely successful,could you please give me more suggestions?

1005452649, 2023-05-04:
must reinstall the torch with cu118

Hezhexi2002, 2023-05-05:
> must reinstall the torch with cu118

But there is no release version of torch with cu118

1005452649, 2023-05-06:
reinstall cuda @Hezhexi2002