ToniRV/NeRF-SLAM #47: TypeError: push_back(): incompatible function arguments. The following argument types are supported:     1. (self: gtsam.gtsam.GaussianFactorGraph, factor: gtsam.gtsam.GaussianFactor) -> None     2. (self: gtsam.gtsam.GaussianFactorGraph, conditional: gtsam::GaussianConditional) -> None     3. (self: gtsam.gtsam.GaussianFactorGraph, graph: gtsam.gtsam.GaussianFactorGraph) -> None     4. (self: gtsam.gtsam.GaussianFactorGraph, bayesNet: gtsam::GaussianBayesNet) -> None     5. (self: gtsam.gtsam.GaussianFactorGraph, bayesTree: gtsam::GaussianBayesTree) -> None

1005452649, 2023-04-26:
Traceback (most recent call last):
  File "./examples/slam_demo.py", line 200, in <module>
    run(args)
  File "./examples/slam_demo.py", line 143, in run
    slam_module.spin() # visualizer should be the main spin, but pytorch has a memory bug/leak if threaded...
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 101, in spin
    output = self.spin_once(input);
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../slam/slam_module.py", line 11, in spin_once
    output = self.slam(input)
  File "/home/user/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../slam/meta_slam.py", line 30, in forward
    output = self._frontend(batch["data"], self.state, self.delta)
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../slam/vio_slam.py", line 119, in _frontend
    x0_visual, visual_factors, viz_out = self.visual_frontend(batch)  # TODO: currently also calls BA, and global BA
  File "/home/user/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 331, in forward
    self.__initialize()
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 655, in __initialize
    x0, rcm_factor = self.update(kf0=None, kf1=None, use_inactive=True)
  File "/home/user/.local/lib/python3.8/site-packages/torch/amp/autocast_mode.py", line 14, in decorate_autocast
    return func(*args, **kwargs)
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 444, in update
    x0, rcm_factor = self.ba(gru_estimated_flow, gru_estimated_flow_weight, damping,
  File "/home/user/linzejun01/linzejun_mutiply_view01/NeRF-SLAM/./examples/../slam/visual_frontends/visual_frontend.py", line 1143, in ba
    linear_factor_graph.push_back(vision_factors)
TypeError: push_back(): incompatible function arguments. The following argument types are supported:
    1. (self: gtsam.gtsam.GaussianFactorGraph, factor: gtsam.gtsam.GaussianFactor) -> None
    2. (self: gtsam.gtsam.GaussianFactorGraph, conditional: gtsam::GaussianConditional) -> None
    3. (self: gtsam.gtsam.GaussianFactorGraph, graph: gtsam.gtsam.GaussianFactorGraph) -> None
    4. (self: gtsam.gtsam.GaussianFactorGraph, bayesNet: gtsam::GaussianBayesNet) -> None
    5. (self: gtsam.gtsam.GaussianFactorGraph, bayesTree: gtsam::GaussianBayesTree) -> None

1005452649, 2023-05-12:
@jrpowers can you help me?

Wqiyu, 2023-10-22:
I have the same question, is it solved?