ToniRV/NeRF-SLAM #48: error occurs when running the demo

Hezhexi2002, 2023-04-26:
@jrpowers Hi,I was struggled into a wired error seems related to gtsam when I run the `python ./examples/slam_demo.py --dataset_dir=./datasets/Replica/office0 --dataset_name=nerf --buffer=100 --slam --parallel_run --img_stride=2 --fusion='nerf' --multi_gpu --gui` demo
the output is shown below:
`Running with GPUs: 0,1
Running pipeline in parallel mode.
I0426 09:02:16.037792 545854 pipeline_module.py:85] Module: VioSLAM - Spinning.
Traceback (most recent call last):
  File "./examples/slam_demo.py", line 202, in <module>
    run(args)
  File "./examples/slam_demo.py", line 145, in run
    slam_module.spin() # visualizer should be the main spin, but pytorch has a memory bug/leak if threaded...
  File "/home/user/high_speed_data/yangshurong/nerf_work/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/home/user/high_speed_data/yangshurong/nerf_work/NeRF-SLAM/./examples/../slam/slam_module.py", line 18, in initialize_module
    from slam.vio_slam import VioSLAM
  File "/home/user/high_speed_data/yangshurong/nerf_work/NeRF-SLAM/./examples/../slam/vio_slam.py", line 9, in <module>
    from factor_graph.factor_graph import TorchFactorGraph
  File "/home/user/high_speed_data/yangshurong/nerf_work/NeRF-SLAM/./examples/../factor_graph/factor_graph.py", line 14, in <module>
    from gtsam import NonlinearFactorGraph as FactorGraph
  File "/home/user/.conda/envs/ai/envs/nerfstudio/lib/python3.8/site-packages/gtsam/__init__.py", line 7, in <module>
    from gtsam import gtsam, utils
ImportError: /home/user/.conda/envs/ai/envs/nerfstudio/lib/python3.8/site-packages/gtsam/gtsam.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN5gtsam25HybridGaussianFactorGraph3addEN5boost10shared_ptrINS_14JacobianFactorEEE
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/user/.conda/envs/ai/envs/nerfstudio/lib/python3.8/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/user/.conda/envs/ai/envs/nerfstudio/lib/python3.8/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "/home/user/.conda/envs/ai/envs/nerfstudio/lib/python3.8/multiprocessing/synchronize.py", line 110, in __setstate__
    self._semlock = _multiprocessing.SemLock._rebuild(*state)
FileNotFoundError: [Errno 2] No such file or directory
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/user/.conda/envs/ai/envs/nerfstudio/lib/python3.8/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/user/.conda/envs/ai/envs/nerfstudio/lib/python3.8/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "/home/user/.conda/envs/ai/envs/nerfstudio/lib/python3.8/multiprocessing/synchronize.py", line 110, in __setstate__
    self._semlock = _multiprocessing.SemLock._rebuild(*state)
FileNotFoundError: [Errno 2] No such file or directory`
I have noticed that there are some similar problems in the issues like [21](https://github.com/ToniRV/NeRF-SLAM/issues/21) and [46](https://github.com/ToniRV/NeRF-SLAM/issues/46),I solved the gtsam build problem according to [21](https://github.com/ToniRV/NeRF-SLAM/issues/21) but still can't get rid of the error I mentioned above,so could you please give me some instructions about the error?looking forward to your reply:-)

Hezhexi2002, 2023-08-31:
> I have the same problem, have you solve this problem?

sorry，but I didn't work on this project anymore therefore

qungvinh1280, 2023-11-13:
Please perform the steps according to this repo: "https://github.com/jrpowers/NeRF-SLAM/tree/master" and it will solve all problems.