ToniRV/NeRF-SLAM #50: wsl2 ubuntu18.04    I don't have a graphical interface, this is the result after running, does it mean success?

lyzhenshuai, 2023-05-25:
[image]
Hello, I'm using wsl2 ubuntu18.04, I don't have a graphical interface, this is the result after running, does it mean success?

zjtde1990, 2023-06-02:
I met the same problem. Do you solve it? How can I get a graphical interface when use "fusion=nerf"?

lyzhenshuai, 2023-06-02:
> I met the same problem. Do you solve it? How can I get a graphical interface when use "fusion=nerf"?

Sorry I didn't solve this problem

ERGOWHO, 2023-09-22:
I have same situation, and same output

rrdongdong, 2023-10-10:
请问解决了吗

ERGOWHO, 2023-10-10:
Sadly no, After this it has a broken pipeline error, this github code has too many bugs, so I turned to use some other papers' work.

Wqiyu, 2023-10-22:
我也遇到类似问题 请问解决了吗

lyzhenshuai, 2023-10-22:
你可试下linux远程桌面，我的远程桌面出问题了，就到这一步就没后续了，在远程桌面里应该有图形界面