ToniRV/NeRF-SLAM #52: run error

liaodaoqing4399, 2023-07-03:
File "/mnt/truenas/scratch/daoqing.liao/NeRF-SLAM/examples/../fusion/nerf_fusion.py", line 301, in fit_volume_once
    self.ngp.frame()
TypeError: 'float' object is not callable

Wqiyu, 2023-10-22:
![U
[image]
ploading 1.jpg…]()
kaz
[image]
a卡在这一步

请问这种情况应该怎么办呢

Wqiyu, 2023-10-22:
有一下两个报错
[image]

[image]