ToniRV/NeRF-SLAM #53: error: droid_backends

northesttiger, 2023-07-15:
ModuleNotFoundError: No module named 'droid_backends'

this error come from "NeRF-SLAM/networks/modules/corr.py", line 4.

this error cannot be solved by "pip install droid_backends'

I think it should have a python file named 'droid_backends.py' in the path "NeRF-SLAM/networks", but it is missing.

anyone else has the same issue ?

MuzeeZ, 2024-01-01:
same problems,have you solved it?

parastoobakhtiarii, 2025-01-14:
have the same error here.
any suggestions?

MuzeeZ, 2025-01-14:
there does have a file named droid_backends.py ,try clone with command “git clone --recurse-submodules <repository-url>" or keep your network stable,by the way,this work have a lot of problems unsolved and doesnt have such good performance as it's claimed.