ToniRV/NeRF-SLAM #62: ''sudo python3 setup.py install'' error occurred

Rosa712, 2023-11-10:
When I run ''sudo python3 setup.py install'' ,error occurred
[image]
it shows I don't have torch module, but the picture shows I have installed the torch

qungvinh1280, 2023-11-13:
> When I run ''sudo python3 setup.py install'' ,error occurred [image] it shows I don't have torch module, but the picture shows I have installed the torch

you can try again with the command: "sudo python setup.py install"

Rosa712, 2023-11-14:
> > When I run ''sudo python3 setup.py install'' ,error occurred [image] it shows I don't have torch module, but the picture shows I have installed the torch
> 
> you can try again with the command: "sudo python setup.py install"

I try again with the command but a weird error is reported. I have installed setuptools
[image]

llxClover, 2023-11-19:
> I try again with the command but a weird error is reported. I have installed setuptools [image]

I also encountered the same error, have you resolved it? I have tested torch.

qungvinh1280, 2023-11-23:
> > > Khi tôi chạy ''sudo python3 setup.py install'', đã xảy ra lỗi, [image]nó cho thấy tôi không có mô-đun đèn pin, nhưng hình ảnh cho thấy tôi đã cài đặt đèn pin
> > 
> > 
> > bạn có thể thử lại bằng lệnh: "sudo python setup.py install"
> 
> Tôi thử lại bằng lệnh nhưng có một lỗi lạ được báo cáo. Tôi đã cài đặt công cụ thiết lập [image]

I have installed the environment for this NeRF-SLAM algorithm many times without encountering any errors like yours. You can follow the steps I have previously performed as follows:
+ Step 1: conda create --name NeRF-SLAM python=3.9
+ Step 2: git clone https://github.com/jrpowers/NeRF-SLAM.git --recurse-submodules
Next, you should follow the subsequent steps as guided in the readme file. 
I will continue to support you if you need it. Good luck!

WN-Wolf, 2023-12-08:
I also encountered the same error, have you resolved it?

WN-Wolf, 2023-12-08:
I also encountered the same error, but I didn't find a useful answer. I tried "python3 setup.py install", but I didn't have permission. So, I added write permission for path /usr/lib/python3.8/site-packages/, then tried "python3 setup.py install" again. At last, it installed successfull.