ToniRV/NeRF-SLAM #67: Is Depth Data used as input?

wangjinhoon, 2023-12-22:
Thank you for providing the code.

I was able to try it out and experience it firsthand thanks to you. Thank you.

The data called Replica seems to be depth data, but when executing slam mode, why is the path of this place given as well? In reading the paper, it seems that only monocular images are used as input, but there is no input data in the folder. Where can I find the input data?

QuangVinh1280, 2024-01-16:
- According to my research, the correct input for NeRF-SLAM is a monocular image. The reason user to input a depth image is to cover the case where the user uses the modified Instant-NGP algorithm (compared to the original repository), without using the --slam parameter in the run command
- As for running directly with your own camera, I believe it is entirely possible if you provide all the following parameters: w, h, fl_x, fl_y. This would require you to write a script to support it.

wangjinhoon, 2024-01-19:
thank your reply

closed 2024-01-19 by wangjinhoon