ToniRV/NeRF-SLAM #69: TypeError: create_empty_nerf_dataset(): incompatible function arguments. The following argument types are supported:

16605506117, 2024-03-12:
Traceback (most recent call last):
  File "./examples/slam_demo.py", line 200, in <module>
    run(args)
  File "./examples/slam_demo.py", line 172, in run
    and (not fusion or fusion_module.spin()):
  File "/22085400519/NeRF-SLAM/./examples/../pipeline/pipeline_module.py", line 88, in spin
    self.initialize_module()
  File "/22085400519/NeRF-SLAM/./examples/../fusion/fusion_module.py", line 25, in initialize_module
    self.fusion = NerfFusion(self.name, self.args, self.device)
  File "/22085400519/NeRF-SLAM/./examples/../fusion/nerf_fusion.py", line 72, in __init__
    self.ngp.create_empty_nerf_dataset(n_images, aabb_scale,nerf_scale, offset,  render_aabb)
TypeError: create_empty_nerf_dataset(): incompatible function arguments. The following argument types are supported:
    1. (self: pyngp.Testbed, n_images: int, aabb_scale: int = 1, is_hdr: bool = False) -> None

Invoked with: <pyngp.Testbed object at 0x7fe31d87c0f0>, 100, 4, 1.0, array([inf, inf, inf]), <pyngp.BoundingBox object at 0x7fe31d8be0f0>

FufenNan, 2024-05-29:
Hi, did u fix this problem?

josyulavt, 2024-08-02:
any update on this please? thank you